r00li/CarCluster #3: PQ25 (Polo 6R Cluster) showing SAFE, gears not being shown

1337Viper42, 2024-03-08:
Hey there, first of all, thank you for this in depth guide, really helped a lot, putting this thing (mostly) together. 
Everything is functioning mostly for me except for gauge for the selected gear. That's not visible at all for me. It just doesn't show anything there. It also shows me "SAFE" above the fuel gauge. That doesn't go away with driving or resetting the cluster either. It basically won't show me the total kilometers driven and I also can't set the digital display for mph permanently on the display. Its just flashing and then disappearing, being covered by "SAFE" again. Maybe you have an idea there. 

Is it possible to make adjustments to the rev counter? Its a bit off for me. On the lower rev band its pretty accurate, but when setting it to lets say 6000rpm, the gauge cluster is displaying almost 7000rpm and at 7000rpm in game, the gauge cluster is already pegged at 8000rpm. Is there a way to fine tune it?

Is it possible to have the speed displayed in finer increments for lets say Beamng? The speedo is rather choppy for me, as in, it only changed the speed in 5kmh increments, making it look rather choppy and jumpy instead of smooth. This is especially noticeable at cruising speeds. 

As for the gauge cluster, its parts number 6R0920860D, date written onto the cluster is 14.02.2010 and it appears to be coming out of a Polo 6R 1.4 petrol. 

Hope you got an idea, if you need to know any more, let me know! 
Thanks in advance!

r00li, 2024-03-08:
Hi!
For the SAFE... have you connected both 12V pins (so 31 and 32)? 
Can you get rid of it by using the trip computer menu controls (pins 21, 22, 23)?
Generally SAFE has to do with immobilizer not being happy. But it's weird seeing it on a Polo cluster of this age.

For the rev counter... In the PQ25.ino you have a value called `rpmCorrectionFactor`. You can change this from 1.0 to something a bit different. Of course this will change the values for the whole range. But will probably get you closer to what you want. Sadly it will probably never be quite perfect.

For the update speed... are you using the USB mode via Simhub or going directly from Beam to ESP32? If you are using Simhub you can increase the serial baud rate in the sketch and simhub settings. If you are going directly then not really. This needs a bit of optimization on my side still.

1337Viper42, 2024-03-09:
Hey there, 
I checked and indeed, pins 31 and 32 are connected to 12V. I also tried getting rid of it via the trip computer menu but without success. 
Do you have an idea why the gears also don't show?

As for the rev counter, I'm gonna give it a shot. This means I have to flash it to the ESP again, right?

And yes, I was using it via WiFi, its still very good and usable, certainly better than my first attempt using an E36 cluster using an Arduino. 

I also found a new issue when playing ETS2 via SimHub. When driving at about 90kmh at 1600rpm, the oil pressure lamp starts blinking and the cluster starts angerly beeping at me. When driving a bit faster or slower it stops, but as soon as you reach this speed again, it starts blinking and beeping after a few seconds.

r00li, 2024-03-11:
I think the gears not showing up has something to do with the SAFE message. But I cannot figure out what. In a real car you would also see this when just applying ignition, but not turning on the car. So it feels to me like it's not getting proper ignition for some reason. Problem with instrument cluster is that they can be coded in many different ways, and that coding impacts how it behaves under certain conditions. 
Do you have the termination resistor installed on your CAN interface? You could try it with or without that and seeing if it helps. Another piece of advice - always plug in the ESP first, wait a second or two and then plug in the cluster. That way the cluster is getting CAN data as soon as it gets powered on.

Yes, you change the value there and flash the ESP with the changes. 

Regarding the oil pressure lamp. You need to connect the "Oil pressure switch" from the pinout. Pin 27 from cluster to pin D15 on arduino. This should get rid of it.

1337Viper42, 2024-03-12:
Did a little bit of testing again but without any change. It does get proper ignition as, when I remove the wire for ignition, the cluster behaves as if you turn the car off and pull the key. As soon as I connect it again, it does its start up routine with all the lamps going on and off again. The SAFE message goes away when I disconnect the ignition pin and comes back after about a second after connecting it again.

So far I ran the CAN interfact without the resistor. Tested again with the resistor but without any difference.

I experimented around with what to plug in first and how long to wait for stuff to happen and for when I do plan on playing, I always plug in the ESP first, wait for around 5-10 seconds and then give the cluster power.

As for the oil pressure situation, I connected all the pins according to the guide to the cluster, the ESP and the CAN interface, so that's all there. Just to make sure I just tried a different cable for the oil pressure switch in case it was damaged, but that also brought no change. It works normal but as soon as you're at or around 1500rpm for more than a few seconds, it starts beeping and flashing. 

I'm currently waiting for the X9C102 to make the fuel gauge also work, hoping that may bring a difference, but I kinda doubt it.

I really do appreciate all the help I'm getting from you, really shows that you care about this as well. So thank you for that.

r00li, 2024-03-13:
I am glad you are making some progress. I find it a bit weird that it would be anything to do with the library versions since the oil pressure thingy is just toggling a pin and nothing else. Could be that you had a bit of a problematic connection and that resolved itself when you switched to the other ESP.

As for SAFE... I have no more ideas for now. My only potential advice would be to try and get a gateway from a same platform car (preferably a similar vintage Polo) and use one of the VCDS clones too see why the cluster isn't happy. And then play around with the coding.

1337Viper42, 2024-03-14:
I do have a genuine version of VCDS though I have no idea how I could manage to connect the cluster to VCDS without a Polo 6R at my hands that I could use for it. 

I've now ordered a Golf 6 gauge cluster (PQ35) as well and I'm going to try if I can make something work with that. In (my) theory the PQ46 (Superb) flash could just work for the Golf 6 cluster as well. Will report on that as well. If something doesn't work, maybe we can work out something together. Thanks for all your patience and help so far!

r00li, 2024-03-14:
In order to use VCDS you will need a gateway (they are usually very cheap). It just needs power, ignition and a connection to the CAN bus with the cluster and your CAN interface. And then you connect the diagnostic CAN to the VCDS. 

For Golf 6 the speed and RPM will most likely work (they are the same across PQ range). The main question are various indicators. Now there are some conflicting opinions if Superb 2 is actually PQ35 (and is an enlarged version of Octavia) or PQ46 (and is an enlarged version of Passat). So the Superb sketch might just work without issues. 

If you want the most hassle free experience I would suggest a Golf 7 cluster though. I think people had a lot less issues with those than PQ platform ones.

kysxe, 2024-03-15:
> I do have a genuine version of VCDS though I have no idea how I could manage to connect the cluster to VCDS without a Polo 6R at my hands that I could use for it.
>
> I've now ordered a Golf 6 gauge [...]

I've also been using a cluster from a Golf 6 GTI and have the same "Key Not Found" issue, which is weird because I've seen people with clusters from the same year and model without this issue and using no CAN interface at all. It could just be something VW added as a "hotfix" on some select models(?) but I'm really not sure, my theory is that it's to do with the immobilizer coil pins not providing some sort of handshake with the cluster as it's not present but I have no idea.

And yes, the PQ46 code is fully compatible with Golf 6 clusters in my experience.

1337Viper42, 2024-03-15:
> I've also been using a cluster from a Golf 6 GTI and have the same "Key Not Found" issue, which is weird because I've seen people with clusters from the same year and model without this issue and [...]

Very good to know that it'll work. Thank you so much for letting me know! The pinout should be the same as in the guide as well, right?
Does the "Key Not Found" message for you bring any issues or does the rest still work as it should? Does the display for the selected gears work for you on your Golf 6 GTI cluster?
I'm very much hoping that I'll get my Golf 6 cluster by tomorrow so I can start experimenting there.

kysxe, 2024-03-15:
> Very good to know that it'll work. Thank you so much for letting me know! The pinout should be the same as in the guide as well, right? Does the "Key Not Found" message for you bring any issues or [...]

No problem :) The pinout is the same for all PQ-style clusters iirc, but [here's a more detailed pinout](https://mhhauto.com/Thread-need-pinout-cluster-MK6-golf) which could help you if you're gonna be experimenting. Unfortunately though, the KNF message takes up the whole display a lot more intrusively than on your Polo cluster and blocks the door-open warning unless you turn off the ignition pin but that's about it, everything else works as it should. I've also included a picture to give you an example on what to expect.

[image]

r00li, 2024-03-15:
> my theory is that it's to do with the immobilizer coil pins not providing some sort of handshake with the cluster as it's not present but I have no idea.

I don't think so. The immo coil is literally just that - a coil of wire, nothing more. 

Weirdly enough my Polo and Superb clusters never complained about the key. Or at least on Superb you can use up the menu buttons to dismiss the alerts. But even then I don't think I had key alerts. 

I am guessing it's more to do with equipment level. It would be interesting to know if any of the cars above had keyless access. That could have some impact on it.

This is a video of my Superb cluster for example:
https://www.youtube.com/watch?v=3lDh-9t-TXc

1337Viper42, 2024-03-18:
Update on my side:
I got the Golf 6 cluster today, just to find that its defective and I essentially got scammed. Really really annoying.
Not even half of the lights of the cluster work, some portions are red, the backlight of the screen didn't work at all and none of the lamps on it worked either. Speed and rpm did work, but it would immediately beep at me. Shining a light at the screen also revealed me the "key not found" which I was unable to clear using the MFA buttons. Though since this thing was defective, I'm not sure if its due to that fact or because it simply doesn't work.

I might buy another Golf 6 cluster next month, we'll see. 

@r00li Depending on how far are you when it comes to implementing MQB features like several indicators as well as the menu controls, I might just get a Golf 7 cluster and  give that a shot instead next month.

r00li, 2024-03-19:
> @r00li Depending on how far are you when it comes to implementing MQB features like several indicators as well as the menu controls, I might just get a Golf 7 cluster and give that a shot instead next month.

I would say Golf 7 is currently quite well supported. What is missing are pretty much the ABS/Traction control lights. Otherwise I think the rest is all there. 

The menu buttons are also included, but this only includes the trip computer. I think the rest of the menu system that you would see is all generated either by the head unit or gateway and not by the cluster itself. This data is then sent to the cluster using BAP protocol, which I wasn't able to figure out.

1337Viper42, 2024-04-04:
I just got a Golf mk7 cluster in the mail that I was able to grab for 25€ and I am excited to report that everything is working! I just haven't tested the fuel simulation yet. But other than that, everything is there and works now. Thanks to everyone! 
[image]

1337Viper42, 2024-04-04:
Tried out more games than just BeamNG and I encountered more bugs with the MQB cluster. Should I open a new issue or just post it here?

r00li, 2024-04-04:
Just post it here, let's see what you found 😄 and if we can do something about it.

1337Viper42, 2024-04-04:
> Just post it here, let's see what you found 😄 and if we can do something about it.

I think this doesn't need much explaining. Speed suddenly drops while accelerating and doesn't come back until I stop again. I noticed this happening at random speeds, but driving at slow speeds (under 30kmh, even for minutes at a time) doesn't cause it. It also doesn't seem to be replated to any gears. Its showing the speed correctly under the Dashboard page and the cluster is still counting the kilometers, it just doesn't display any speed. This happened using ETS2 via SimHub. It didn't happen with BeamNG using the in-game Out-Gauge feature. 

https://www.youtube.com/watch?v=Sxp1CYNqmG4

r00li, 2024-04-04:
> I think this doesn't need much explaining. Speed suddenly drops while accelerating and doesn't come back until I stop again. I noticed this happening at random speeds, but driving at slow speeds [...]

Oh dear. Not again with the speed drops. These have been a constant with VAG clusters, and every time I resolve it, it pops up again somewhere 😄 . 

I do find it weird that, it only happens with Simhub, but not directly from Beam. That is slightly weird.

Can you maybe have a look at this PR, and try the version before the changes from this PR were merged?
https://github.com/r00li/CarCluster/pull/1/files

1337Viper42, 2024-04-05:
Unfortunately this just made it worse. Its now also happening with Beam.

r00li, 2024-04-05:
Kind of expected. So, at least we know that the PR helped 😄 . 

Which game is running should have no impact on this. Does this happen with Beam if you are using it via simhub as well?

If it does... did you by any chance change the baud rate in the sketch from the default 115200? If not... try using 921000. If yes, then try going back to 115200. I am wondering if the ESP is struggling with keeping the transmission rate when working over USB.

1337Viper42, 2024-04-05:
It was set to 115200 and it did the same in Beam via SimHub. I now set it to 921000 and well, that did the trick! At least for Beam via SimHub. Haven't tried ETS2 yet. Will do in a few days as my wheel bit the dust earlier today. Looks promising so far! The speed of which rpm and speed changes also increased drastically compared to doing it via WiFi and Outgauge Support. Its now refreshing much quicker and in much smaller increments which is just an absolute treat. Thank you so much for all the help! 

What do you reckon, how big of a job is it to make other indicators/lamps like ABS and so on work on MQB as well now?

kysxe, 2024-04-06:
> I think this doesn't need much explaining. Speed suddenly drops while accelerating and doesn't come back until I stop again. I noticed this happening at random speeds, but driving at slow speeds [...]

The speed-drop fix added an adjustment to the timing within the distance calculation and thus the message frequency in Simhub should be set to 20Hz instead of the default 60Hz. Although, I find it interesting that increasing the baudrate seemed to do the trick for you so maybe it was a transmission issue after all? Regardless I'm glad you got it all working :)

1337Viper42, 2024-04-07:
> I am glad it's working now. As for indicators... I think most of them are there. The ABS/traction I think is the biggest missing piece. However I don't know the CAN ID for it. I am building a full on Golf 7 sim rig for myself, so I will try again in the future. But I need somebody who knows the ID.
> 
Can I somehow help you with this? While I'm mostly clueless on how coding and stuff like that works, I am willing to learn and one of my cars (my daily) is a 2016 Golf 7 GTD Variant, so I might be able to do different things than just fiddle around with a gauge cluster and no car around it. I also have a genuine copy of VCDS of that helps at all.

r00li, 2024-04-07:
Sadly the only way to actually get this IDs is by taking out the cluster and basically hooking up a CAN bus interface to the line (preferably without disconnecting it - since that would cause errors) and then recording the stream while triggering the indicator that we need.

1337Viper42, 2024-04-07:
Let me know what I need and how to record stuff. I could in theory just use a breakout cable for the OBD port, get the CAN signal from there and just trigger the lamps manually by using VCDS via a segment test. Now I'm not sure if the segment test for the Golf 7 via VCDS will trigger individual lights or just enable them all at once. But in theory, that would you leave you with a bunch of IDs that you could individually test and label.

r00li, 2024-04-07:
No, you can't use the OBD port for that. The data on the OBD port is filtered and won't show what we need. You need to probe on the actual CAN lines going to the cluster. 

Likewise the segment test will just tell the cluster to do the test, not send the CAN messages that would normally light up those lights.

Walfrido22, 2024-04-08:
Good morning I figured out which pins would be used to change the options on the on-board computer, for example my golf mk7 cluster is in the fuel consumption part, I would like to put the digital km/h, and is that possible?

[http://www.sifc.com.br/template/img/banner-certificado.jpg]

<http://www.sifc.com.br/>

1337Viper42, 2024-04-08:
You can just cycle through the display via the ESPs Dash. Just connect to the IP of your ESP and then you can go through the menus and set it to display the digital speed.

Walfrido22, 2024-04-28:
Hello everything is fine ? I was never able to establish the ESP 32 connection with my internet, I find the carcluster network, I type the carcluster password, I select my wifi network, I enter the password and it doesn't connect after a while the car cluster network disappears, how can I solve this I play plus ets2 via simhub but I would like to make this connection to change some things, thank you very much in advance for your attention

<http://www.sifc.com.br/>

[http://www.sifc.com.br/template/img/banner-certificado.jpg]

<http://www.sifc.com.br/>

r00li, 2024-04-29:
@Walfrido22 That is an issue with the wifi of the ESP32 being quite bad. If possible try moving closer to the wifi router/access point. Usually that resolves the issue. 

If that doesn't help then you will need to play with the wifi settings on your router/access point a bit. Try googling for issues like "ESP32 wifi not connecting" or similar.

1337Viper42, 2024-06-09:
@r00li Hey there, I watched this CAN Bus seminar recently and they mentioned this Open DBC database. Would this perhaps be of any help for you for further implementations of functions with the MQB clusters as an example?
I'm just gonna drop you the links to both the seminar and the database here.
https://www.youtube.com/watch?v=j8MXMMqgksk
https://github.com/commaai/opendbc

r00li, 2024-07-01:
@1337Viper42 Sorry for forgetting to respond earlier. Some of the stuff here was actually developed with some trace and dbc files from CommaAi/Opendbc. So, I am aware of it and do check it when working on stuff.

RedSmallVest-863, 2025-08-25:
> > Hi, I also have a Golf 6 instrument panel. Interestingly, I also encountered the problem of no backlight, but everything else seems to be normal without beeps, including the ABS light and ESP light. Everything works normally.
> 
> Hi! Regarding the backlight... you might want to try this PR - I think Golf 6 might be closer to PQ46/Superb than Polo: [#39](https://github.com/r00li/CarCluster/pull/39) So, switch to PQ46, and try the changes from there. I am finally back and I will try and get that PR merged soon.

Thank you very much for your help. I followed your instructions and successfully solved the problem. I appreciate if you can help to advise if there is any new progress on the MQB cluster issue for the new model?  Thanks again sincerely.