r00li/CarCluster #4: Does BMW E60 cluster work?

Betty1909, 2024-03-12:
Can it work with BMW E60 cluster?
I have change to 100kbp and it success to control backlight, but rpm and speed
not work.

And backlight flash again and again.

Thanks!

r00li, 2024-03-13:
Hello!
Not for now. E series is still in the works. But it is next on my list. I have the cluster already and will start working on it soon-ish. Probably a few weeks. I'll let you know when it's ready.

Betty1909, 2024-03-14:
Cool,
Thanks for your big work. 😀
Can't wait to playing with it!
[image]
Here is my cluster, only rpm work. lol

r00li, 2024-08-05:
Hello! I am happy to report that BMW E60 is now finally supported. Currently the support is still a bit basic, but expect more to come at some point.

closed 2024-08-17 by Betty1909