r00li/CarCluster #5: A DTC shows on BMW F10 cluster

Betty1909, 2024-03-29:
Everything works good, but still have a DTC on it.
Looks like turning system module maybe power steering?
[image]

And a little problem.
It'a good when only RPM needle runing, but both speedo needle and rpm
rpm needle goes a little lag (few).

https://github.com/r00li/CarCluster/assets/29305277/61a8c85e-e6d5-44fd-96a6-2fab167e28aa

Sorry for my bad english. :)

Betty1909, 2024-03-29:
And this is my PCB, includ two x9c102 and mcp2515, 2551. :))
[image]

BrayanOfficiel, 2024-03-29:
> And this is my PCB, includ two x9c102 and mcp2515, 2551. :)) ![IMG20240326215124](https://private-user-images.githubusercontent.com/29305277/317970875-0dd8ae4c-2d80-4347-9069-694040e53921.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE2OTgwNDgsIm5iZiI6MTcxMTY5Nzc0OCwicGF0aCI6Ii8yOTMwNTI3Ny8zMTc5NzA4NzUtMGRkOGFlNGMtMmQ4MC00MzQ3LTkwNjktNjk0MDQwZTUzOTIxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI5VDA3MzU0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTliNzdhNzlhMmIzNGRhZTcyMjMzNDc2NWRmYzQ2N2EyNmY1ZTAwM2M5NGMzNjM2NGYxYzBjMmU5NGY0YjkxMTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.nchYujrnWdoSVOjUN7Ek629LElLM-dFL--rNXu2YwXM)

For the lag, try setting the baud rate to 921000, it also happened to me while playing on serial

BrayanOfficiel, 2024-03-29:
> Everything works good, but still have a DTC on it. Looks like turning system module maybe power steering? ![IMG20240327194136](https://private-user-images.githubusercontent.com/29305277/317967935-5e108103-706a-4c08-bcc4-15fe32c4b546.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE2OTgxOTgsIm5iZiI6MTcxMTY5Nzg5OCwicGF0aCI6Ii8yOTMwNTI3Ny8zMTc5Njc5MzUtNWUxMDgxMDMtNzA2YS00YzA4LWJjYzQtMTVmZTMyYzRiNTQ2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI5VDA3MzgxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU5YzQ4MmJiYzU5Nzc4Yzc5M2VhYjM1MjgwNDgxMjdiMTQ5MDI4MTM5M2I4ZmExYTlhZDU4MDgwNzA1NWQyNjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Ui8gDD1DKwOmdEQ8J7Sx-DIS7jhYlAjIXg-dDH9i85g)
> 
> And a little problem. It'a good when only RPM needle runing, but both speedo needle and rpm rpm needle goes a little lag (few).
> 
>  VID_20240329140220.mp4 
> 
> Sorry for my bad english. :)

For the lag, try setting the baud rate to 921000, it also happened to me while playing on serial

r00li, 2024-03-29:
Yep, if you are using the Simhub connection via USB, then increating the baud rate should help. Sadly I didn't have a chance to update the sketches for that yet.

As for the DTC error. I get the "Cornering ability restricted" error as well on my F10 cluster. Sadly I cannot find a solution for that. I think this is due to a differences in coding of the cluster - this was made by using a log from a real F10 car, and even when replaying that whole log the issue is still there. So I am guessing it's either manufacture date or equipment level specific. Also, some people don't get that error. I guess the only way around this is to find the CAN ID responsible for it.

KMC-Workshop2024, 2025-12-15:
> Yep, if you are using the Simhub connection via USB, then increating the baud rate should help. Sadly I didn't have a chance to update the sketches for that yet.
>
>
>
> As for the DTC error. I get the [...]

I could give you some info about the DTC. It's the fault code of AirSuspensionSystem, maybe the module of AirSuspension is always sending CAN messages to the wires, if you try to add the code from Air Suspension System Module to the ESP32, the problem may be solved?