r00li/CarCluster #6: Unable to turn on F10 cluster

wyv14, 2024-03-29:
Hello there. 
I have followed the wiring from ESP32 to a CAN BUS module, and also +12v/GNF/CAN H/CAN L of the F10 cluster connector.
(I am able test the CAN BUS module loop test from IDE-Example-mcp can example).

However, after compiling, and setting up the DASH web interface, I am unable to control the F10 cluster. The cluster is still in sleep mode. 
(pressing the odo button wakes up but sleep again after few secs)

_Question.. The DASH able to control the cluster, am I right?_

If yes then it could be the cluster or the CAN BUS module?
Is there anyway to isolate where went wrong?

Thanks!

r00li, 2024-03-29:
Hi!
So you are saying that the loopback example from the MCP library works?
Did you make any changes to that example or did you run it as is?
I am asking because the loopback example uses pin 10 for CS line, while my library uses pin 5 for that. So if loopback works, then try changing SPI_CS_PIN to 10 in the code here.

The DASH page should be able to control the cluster, yes. Even just connecting the ESP to power and wifi (without doing anything on the dashboard) should wake up the cluster. 

Seeing how you are not even getting ignition then the issue is most likely either in your wiring or your ESP32 board/pin configuration.

wyv14, 2024-03-29:
Hi Andrej, 

thanks for replying. 
Yes I did changed the pin accordingly.
Have disconnected and connected the wiring a few times and end up with same issue :(

Just 1 more question. 
When you connect the cluster pin 11 which is the wake up signal with +12v, the cluster will be in sleep mode right?
It's will only wake up when the DASH web interface is up (as DASH by default the IGNITION is ON which will send the signal regularly). 

Thanks again!

Betty1909, 2024-03-29:
Check all wiring, 
Unplug +12v for F10 cluster (keep esp32 power, CAN H/L, Gnd), and wait for 20sec (make cluster shutdown).
Replug +12v to F10 cluster, and it should wake up and working.

wyv14, 2024-03-29:
Thanks Andrej & Betty.

Betty's tip helps. In the past, I would power up the cluster first, then either debug/compile on the ESP. 
This time round, I let ESP status up first, then power the cluster. (the cluster is via a power adapter thus just a matter of switching ON & OFF)

I would like to further ask that from the DASH setting, the speedo and RPM fluctuate up and down. Is this normal?

https://github.com/r00li/CarCluster/assets/163303243/20a697fd-4781-4db7-b9e7-bdb6ee716064

[image]

Betty1909, 2024-03-29:
Nice to meet you!
Where are you from?
I'm from Taiwan.

wyv14, 2024-03-30:
Hi Betty. 
Nice meeting you too. 
I am from Singapore. 
The cluster is given by my brother who lives in Taiwan, Xindian. 

And just nice found Andrej's blog, and learning this fantastic project of his ;)

r00li, 2024-03-30:
> I would like to further ask that from the DASH setting, the speedo and RPM fluctuate up and down. Is this normal?

No, it is not. Usually this happens because CRC or counter is wrong for one of the CAN IDs. But I have no idea. why this is happening for you. Does this also happen when using it with a game?

wyv14, 2024-04-02:
I have not used it with a game yet.. Will update again, thanks!