r00li/CarCluster #8: The speed on the dashboard drops to 0 km/h

frangomez2908, 2024-04-22:
Hello!, I have a dashboard of a VW Polo 6r (PQ25) I have managed to carry out the entire process and it connects perfectly, but the same thing happens to me as in the demonstration video, the speed drops to 0, and when I stop the vehicle and come back to accelerate it rises again, until the process is repeaHello!, I have a dashboard of a VW Polo 6r (PQ25) I have managed to carry out the entire process and it connects perfectly, but the same thing happens to me as in the demonstration video, the speed drops to 0, and when I stop the vehicle and come back to accelerate it rises again, until the process is repeated. (About 7 seconds). I have seen that with other dashboards you have managed to solve it, but in this one I have not seen a way, any ideas? Thank you!ted. (About 7 seconds). I have seen that with other dashboards you have managed to solve it, but in this one I have not seen a way, any ideas? Thank you!

aslobodskoy, 2024-04-22:
I've some issues with ESP32 stability on high frequency but it work perfectly for me on 80MHz

frangomez2908, 2024-04-22:
I have lowered it to 80Mhz but it remains the same, do you have all the cables connected as the creator indicates?

aslobodskoy, 2024-04-23:
@frangomez2908 I've got MQB dashboard from Golf7

r00li, 2024-04-23:
Generally the speed drop should be fixed on the PQ25 clusters as well. See the second part of this video: https://www.youtube.com/watch?v=3lDh-9t-TXc

Are you using it with Simhub? If yes, then try increasing the baud rate in the sketch and simhub to 921000.

frangomez2908, 2024-04-23:
I am testing it both on the dash server, on forza horizon 4 (Wifi) and on simhub with the speed at 921600, in all cases the speed drops to 0 after 10 seconds, however the revolutions work great. I can't think of what else to do.

Patrick-Cunningham, 2024-04-27:
Try adjusting this line https://github.com/r00li/CarCluster/blob/main/Dashboard%20sketches/PQ25/PQ25Dash.cpp#L105 from 146 to 148

frangomez2908, 2024-04-28:
I have tried doing it from 146 to 148 and the result is the same, I don't know what could be happening

r00li, 2024-04-29:
Yeah, I think @Patrick-Cunningham is on the right track here. Play around with this value a bit more. Also try the line just below it:
`distanceCounter += speed * dashboardUpdateTime50 / 182` 
Try changing the 182 to something else. Not by much though. These parameters were derived pretty much by experimentation, and while they work for my clusters, it seems like some aren't quite happy with it.

Patrick-Cunningham, 2024-04-30:
I have 182 set to 72, give that a shot, the dropping of the speedometer is caused by a correlation error between the speed provided and the distance. I have a US Spec Golf pq35 cluster and this has been working flawlessly for me in my project

frangomez2908, 2024-04-30:
Thank you all, in the end I edited the program and sent the time data via serial port so I wouldn't be uploading the program all the time. I discovered that I have the same setup as you. From 60 to 80 it works correctly for me. I left it at the end at 70 to have a middle ground. Thank you so much

closed 2024-05-13 by r00li