r00li/CarCluster #9: How to generate CAN wakeup signal

veso266, 2024-04-29:
Hello there, I have a very wierd question

I have a test bench for PQ platform which you can see on the picture: https://imgur.com/fyGPqXu

On my Bench is RNS510 and 3C0 907 530 L gateway module

I would like to simulate ignition message
I use MCP2515 module with my Arduino (seerduino the only difference is that u can also put it on 3.3V operation instead of 5V)

I decided to use Instroment cluster (because I knew my RNS510 uses MFD display to send radio station information to instrument cluster so it has to have connection to this blue cable) blue cable to hook inside my CAN Netowrk

I also discovered that VW PQ cars have 3 canbuses: https://imgur.com/a/FAXZZzY
Infotaiment - 100kbps
Conveniences - 100kbps
Drive train - 500kbps

I tried to receive some packets and discovered that packets were only uvailable on 500kbps drive train bus, 100kbps bus didn't have any data, but I guess ignition message should be on drive train bus anyway, and gateway would take care of forwarding it to RNS510 at the right speed

Since I sadly have no acsess to any PQ vehicle anymore so I don't have any real data

I did fine tune your PQ45 example, to maybe at least try to send speed to my RNS: 
[PQKlemme15GeneratorAttempt.zip](https://github.com/r00li/CarCluster/files/15150023/PQKlemme15GeneratorAttempt.zip)

but nothing happens, if you look at this picture: https://imgur.com/PAH2PBI
RNS510 clearly doesn't see the speed packet 

I know here many things can go wrong, my packet is maybe not correct, maybe I am using the wrong port to feed my CAN message to
I also have acsess to radar port on my lovly harness, but since its CAN this shouldn't matter, the way I understand this is that as long as the messages reach the gateway, then it should forward them to the right module

So I am just wondering if you maybe have some CAN data from PQ platform cars somewhere or would know if it matters to what point in the CAN network you feed your data to

Thanks for Anwsering and Best Regards

r00li, 2024-04-29:
I didn't do anything similar with PQ platform, but I did do it with MQB:
https://www.youtube.com/watch?v=VAaX5-TPA4Q

At least on MQB the data for the headunit is actually coming from the instrument cluster. VW uses a protocol called BAP, which is transmitted as a single CAN ID. So, cluster gets the speed data, then sends it back to the gateway as a BAP message, and then that gets transmitted by the gateway to the infotainment CAN.

So it does matter where you inject your data. Now what you see in the video is just a testbench setup. But what I am doing in some later stages is actually using two CAN modules -> intercepting the messages from gateway to instrument cluster on one interface, then faking the data and sending it back out to the cluster on the other interface. For the other direction (cluster to gateway) I am just passing data through directly. And on my headunit the average speed and consumption work correctly. 

Now this is based on MQB. Could be that PQ is a bit different. I know that on MQB, cluster is connected to the convenience CAN, while if I am not mistaken on PQ it is drivetrain.

veso266, 2024-04-29:
I see 

So the way I understand this is that if I inject data into the blue connector (instrument cluster) no matter what I do it will never reach the gateway (or even if it reaches the gateway it will discard the message, despite being correct)? 

My ultimate goal is to try to send the proper Klemmen15 message so I could tell the RNS and other CAN modules on the network that the CAR is running 

Simulating speed and proper VIN would also be a neat bonus 

I have this harness: https://www.aliexpress.com/i/3256801217104176.html?gatewayAdapt=bra2glo4itemAdapt 

Maybe the radar connector would be a better candidate to inject my wakeup messages to? 

Or the OBD2 port? 

I always thought the the advantage of CAN is that no matter where u feed ur data, it would always found its way to the right module 

But I am begining to realize its not that way
BTW: to what gateway pins u connected ur BCM (this is the device thats generating wakeup messages right?), gateway pinout hopefully didnt change between PQ and MQB platforms

r00li, 2024-04-30:
> So the way I understand this is that if I inject data into the blue connector (instrument cluster) no matter what I do it will never reach the gateway (or even if it reaches the gateway it will discard the message, despite being correct)?

It's a bit more complicated than that. You need to inject your CAN messages on the bus where they would actually appear.

> Simulating speed and proper VIN would also be a neat bonus

I suggest you very much avoid sending the VIN packet under any circumstances. Only bad things happen with that 😄 .

> I have this harness: https://www.aliexpress.com/i/3256801217104176.html?gatewayAdapt=bra2glo4itemAdapt
>
> Maybe the radar connector would be a better candidate to inject my wakeup messages to?
> 
> Or the OBD2 port?

Definitely not the OBD2 port. That is pretty much useless for any project like this. Data on there is completely filtered and you only get the diagnostic messages on there. Nothing actually really relevant for this purpose.

>  BTW: to what gateway pins u connected ur BCM (this is the device thats generating wakeup messages right?), gateway pinout hopefully didnt change between PQ and MQB platforms

If I remember correctly BCM goes on the convenience CAN on MQB. And, yes. BCM is the one generating the ignition message. At least on MQB. Although on PQ I know that ignition signal originates with the steering column electronics control unit. But that is a wire.

> Now I just have to figure out what packet I have to send so gateway will tell RNS510 the car is running

Good question. Unlike MQB, PQ platform uses ignition by pin (instead of CAN) for the instrument cluster, so I never looked into what exactly is the ignition packet on PQ. Question is if there even is one.

r00li, 2024-05-13:
Yeah, definitely do check the timings. Some components can be quite unhappy with incorrect timings.

For MQB I never tried simulating the gateway sadly. Usually for this project I deal directly with clusters. And for the other project I have the BCM/Gateway/Lock/Cluster so I can get the proper ignition signal.

Yeah, on PQ you are probably safe (unless dealing with Audi) but on MQB I had a cluster go into component protection before and I could never recover it. 
I don't think the gateway itself would lock. But things like cluster and radio definitely. Again, probably not on PQ, but still, better to be cautious.

MIGINC, 2024-06-14:
To get my MIB STD2 PQ to see that terminal 15 is active, I need to send it ID 0x575 - 0x47, 0x20, 0x00, 0x20 
[image]