r00li/CarCluster #13: Gear Display

CreeBoom2020, 2024-08-11:
Hi i was wondering if u figured out how to display the curent gear on the bmw f cluster if no then i can help u out with that!(and i ment gear and not just D,R,N,...).

r00li, 2024-08-17:
Hi! 
As far as I am aware on automatic clusters with regular transmission that is not possible. But if it is and you know how, then let me know and I will see if I can add it.

CreeBoom2020, 2024-08-17:
hey!
so if u take this     unsigned char transmissionWithoutCRC[] = { counter4Bit, selectedGear, 0xFC, 0xFF }; //0x20= P, 0x40= R, 0x60= N, 0x80= D, 0x81= DS 
and change the first byte to 0x10 then it wil display gear 1 but shortly because you have to add one until u are at for gear 1 exaple 0x19 then u reset to 0x10 thats basicly it. and just add 10 for each gear 1 = 0x10 2 = 0x20  3 = 0x... and for n u just go back to counter4Byte! very simple! :] gona past my code in here so u can take that and add it: :]

CreeBoom2020, 2024-08-17:
int manual = 0;
       int gmax = 0;
       int runs = 0;
       int runsg = 0;
       int runsb = 0;

gmax += 1;
if (gmax == 0x09) {
gmax = 0;
}
    
selectedGear = 0x00;
manual = counter4Bit;

if (gear == "1") {
manual = 0x10;
runsg = manual;
manual += gmax;
runs = 0;
} else if(gear == "2") {
manual = 0x20;
runsg = manual;
manual += gmax;
runs = 0;
} else if(gear == "3") {
manual = 0x30;
runsg = manual;
manual += gmax;
runs = 0;
} else if(gear == "4") {
manual = 0x40;
runsg = manual;
manual += gmax;
runs = 0;
} else if(gear == "5") {
manual = 0x50;
runsg = manual;
manual += gmax;	
runs = 0;
} else if(gear == "6") {
manual = 0x60;
runsg = manual;
manual += gmax;	
runs = 0;
} else if(gear == "7") {
manual = 0x70;
runsg = manual;
manual += gmax;	
runs = 0;
} else if(gear == "8") {
manual = 0x80;
runsg = manual;
manual += gmax;
runs = 0;
} else if(gear == "9") {
manual = 0x90;
runsg = manual;
manual += gmax;
runs = 0;	
}

if(gear == "N" or gear == "0" or gear == " " or Game == "ETS2") {
runsb = runsg;
runsb += gmax;
manual = runsb;

runs += 1;

}

if(runs > 3) {
manual = counter4Bit;
}

if ( gear == "R") {
selectedGear = 0x40;
manual = counter4Bit;
}
    unsigned char transmissionWithoutCRC[] = { manual, selectedGear, 0xFC, 0xFF }; //0x20= P, 0x40= R, 0x60= N, 0x80= D, 0x81= DS
    unsigned char transmissionWithCRC[] = { crc8Calculator.get_crc8(transmissionWithoutCRC, 4, 0xD6), transmissionWithoutCRC[0], transmissionWithoutCRC[1], transmissionWithoutCRC[2], transmissionWithoutCRC[3] };
    CAN.sendMsgBuf(0x3FD, 0, 5, transmissionWithCRC);

CreeBoom2020, 2024-08-17:
that soudl be the code let me know if u get i forgot something! and good night! :]

CreeBoom2020, 2024-08-17:
[image]
[image]

CreeBoom2020, 2024-08-17:
and also sometimes the sos and tirepresure malfunktion came on also fixed that and added suport for dors trunk frunk open or closed and engine overheated check engine and tire presure low when poped in beamng and efficency dynamics :] so can get u the ids or bytes for that aswell!

r00li, 2024-08-17:
Thank you for this. I will try it out when I come back from my summer vacation.

Getting rid of the errors would be nice yes. The SOS one especially. 
Any chance you know how to get rid of the "cornering ability restricted" error? It's the only one that I have on my F10 that I can't seem to get rid of.

CreeBoom2020, 2024-08-18:
Hey so i Put my Code online a few houres ago u can Just Look in to that and Look what i cooded in there to do each thing! Have fun on ur vacation!
https://github.com/CreeBoom2020/Bmw-F3x-6wa-cluster-with-simhub-for-beamng

CreeBoom2020, 2024-08-22:
hello r00li,
do u still have the f31 in a box? if yes can u send me a log of the mfl butons? that woudl be help full. thx in advance

r00li, 2024-08-22:
I do, yes. Will try getting it to you as soon as I can. But will probably take a few weeks.

CreeBoom2020, 2024-09-10:
hey good day r00ly,
did u get to record the can bus jet or did u find no time for it yet?

CreeBoom2020, 2024-09-27:
Hi,
Thanks! and just one more question how is the Klimate control pannel hooked up? to the radio control panel and then k can2 or lin?
THX

closed 2025-04-11 by CreeBoom2020