r00li/CarCluster #15: BMW F30 Cluster shows "Front-End Collision warning failure" and some other problems occur

NullFreund70254, 2024-08-31:
Hi there,

On my BMW F30 cluster, when ignition is on, it always shows "Lane Assistant Failure" and "Front-End Collision warning failure".
Normally it doesn't show that, but I only found the can bus code for the Lane Assistant and put it in your program:

`unsigned char LaneAssistWithoutCRC[] = { 0x22 | counter4Bit, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};
unsigned char LaneAssistWithCRC[] = { crc8Calculator.get_crc8(LaneAssistWithoutCRC, 7, 0x28), LaneAssistWithCRC[0], LaneAssistWithCRC[1], LaneAssistWithCRC[2], LaneAssistWithCRC[3], LaneAssistWithCRC[4], LaneAssistWithCRC[5], LaneAssistWithCRC[6] };
CAN.sendMsgBuf(0x327, 0, 8, LaneAssistWithCRC);`

Atleast the Lane Assistant Error went away, but I don't know if i like it, that the Lane Symbols on my cluster keep on flashing.

Not only that is a problem, somehow my rpm stays only on "ready"/never moves from it's place, gears showing on the cluster are randomly spawning, and then going away and the check engine light is on(must be because of the Front-End Collision warning failure)
I tried out this code from another guy who made a simhub program especially for the F30 Cluster: [https://github.com/CreeBoom2020/Bmw-F3x-6wa-cluster-with-simhub-for-beamng](url) , but it still didn't work. 

That's why my question is, do you have by chance the correct bytes for the Lane Assistant and the can bus code for the Front-End collision warning message?

Thanks

r00li, 2024-09-03:
Hi!
I will look into adding the lane assist code since you have given me the ID. Here is a snippet from a CAN log from an F10 that can maybe be of some use to you as well:
[image]

Now for the RPM... you are the second person to say that the RPM doesn't work with an F30 cluster. I am not sure why. The same code works for my F series mini, my F10 and my F30. Do you know if your cluster comes from a diesel/gasoline cars and is it automatic or manual? Also, is it a basic cluster or 6WA?

As for the front end collision warning... If you can find the ID I can see if my log has that message in there. Sadly I don't know which ID is responsible for it. And since my donor cars didn't come with that feature I am not getting any errors to check for myself.

NullFreund70254, 2024-09-03:
Hey, thank you so much for the Lane Assist Can Id!!! 

Now my question has been answered, it indeed uses only 4 bytes, and not 8, lol. Seems to be that the 1st byte starts from 50 and ends at 5E, while the other 3 bytes stay in the same position. Somehow it was hard for me analyzing the can ids from a F10/F30 because i don't have a Log file nor do i have a F chassis BMW, I have a G26 and it is definitely useless. Yeah i needed to pray on the counts, randoms and what i find in the internet.😐 Heard also that the Checksum on the F30s are shittier than the ones in the f10, why? I don't know.

My cluster is from the European market, it comes from a 325d(so diesel) and it is automatic, that is what the private seller told me after replacing his cluster for a 6wb. Oh yes, the cluster which I have is a 6wa cluster. 
Seeing from your f30 cluster photo in the Readme file, yours is a manual, or?

And just for your Information, the Cluster test in the hidden menu on my Cluster is successful, so there are no problems with the Motors in the cluster. Seeing from your photo in the Readme file, yours is a manual, or?

One more thing, I realized something. Before the Front-End collision warning occurs, so like around 2-4secs where the check engine light doesn't pop up after sending ignition, sometimes the RPM and the Gears work, but then, after the Check Engine Light comes and the Front Collision Warning pops up, they go back to Ready. I basically mean, that i turn on my cluster while the car ingame is running, rpm and gears are showing, then after 4secs, the errors come again and RPM goes back to Ready and Gears disappear.

r00li, 2024-09-06:
Ooof, never heard this one about F30 being harder. It shouldn't be the case. Unless they changed some things with various facelifts.

Mine is a UK spec (because they are cheaper in the EU for some reason). And, yes. It is from a manual. Actually it might not have been, but I also have a FEM and I recoded it to a manual one. 

That is interesting. And also quite promising. It makes me think that the IDs that we are sending are correct, and it's probably something that is missing which is causing the issues. Problem is of course finding out what.

NullFreund70254, 2024-09-06:
Oh, yeah, this is one important thing. I heard in other Discord servers, that the checksum in Automatic F30 clusters is much harder than on the manual F30 ones. That's good that you got it coded to a manual, I think that it might be the problem, because my cluster is Automatic and yours is manual. I think it's not only on the F30 a problem. Might be that other F Chassis models have the same problem. Sucks. But this could be my Speculation.

NullFreund70254, 2024-09-07:
Oh, and one question, does it work, that you can program the bench from a us model to a europe model?

r00li, 2024-09-08:
Never had any issues with my F10 - which is also automatic. Not sure about the others.

As for EU-US models... no idea. If they are using the same part numbers then you should be able to code them. If they are different then it really depends on the software that they have.

KMC-Workshop2024, 2025-12-12:
> Hi there,
> 
> 
> 
> On my BMW F30 cluster, when ignition is on, it always shows "Lane Assistant Failure" and "Front-End Collision warning failure".
> 
> Normally it doesn't show that, but I only found the can bus code for the Lane Assistant and put it in your program:
> 
> 
> 
> unsigned char LaneAssistWithoutCRC[] = { 0x22 | counter4Bit, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};
> 
>  unsigned char LaneAssistWithCRC[] = { crc8Calculator.get_crc8(LaneAssistWithoutCRC, 7, counter4Bit), LaneAssistWithCRC[0], 
> 
>  LaneAssistWithCRC[1], LaneAssistWithCRC[2], LaneAssistWithCRC[3], LaneAssistWithCRC[4], LaneAssistWithCRC[5], 
> 
>  LaneAssistWithCRC[6] };
> 
>  CAN.sendMsgBuf(0x327, 0, 8, LaneAssistWithCRC);
> 
> 
> 
> Atleast the Lane Assistant Error went away, but I don't know if i like it, that the Lane Symbols on my cluster keep on flashing. Yes, i know that i am using count.
> 
> 
> 
> Not only that is a problem, somehow my rpm stays only on "ready"/never moves from it's place, gears showing on the cluster are randomly spawning, and then going away and the check engine light is on(must be because of the Front-End Collision warning failure)
> 
> I tried out this code from another guy who made a simhub program especially for the F30 Cluster: [https://github.com/CreeBoom2020/Bmw-F3x-6wa-cluster-with-simhub-for-beamng](url) , but it still didn't work. 
> 
> 
> 
> That's why my question is, do you have by chance the correct bytes for the Lane Assistant and the can bus code for the Front-End collision warning message?
> 
> 
> 
> Thanks

I guess that code the Cluster to disable the LaneAssist may by useful?