r00li/CarCluster #16: F10 Cluster not working in simhub

karrui, 2024-09-04:
DASH works when I toggle the buttons there, but it does not seem to work with SimHub serial device. Am I doing something wrong?

[image]

This is the computed values I get from inside Euro Truck Simulator, which does show some values. The cluster is not responding to these values though.
[image]

karrui, 2024-09-04:
is it my usb port? the serial data is a bunch of gibberish.

[image]

r00li, 2024-09-06:
Hi!
Sorry it took me a while to respond...

Your settings look correct. Are you sure that COM5 is your ESP32? Do you have any other devices connected to the computer that might show up there?

Are you using the latest code from here? In a slightly older version of this code I was using a slower baud rate. Check in the code that you have the speed set correctly. Find the line that says `Serial.begin` the number in brackets should be the baud rate that you have set here.

karrui, 2024-09-07:
Yup! Correct baudrate. I'm using ESP32 with a CH340 chip. That might be the issue. I've got a new ESP32 on the way with the CP2102 chip instead.

r00li, 2024-09-08:
I don't think that should matter.
Can you open Arduino serial monitor at the correct baudrate and send this:
`{"action":10, "spe":54, "gea":"2", "rpm":3590, "mrp":7999, "lft":0, "rit":0, "oit":0, "pau":0, "run":0, "fue":0, "hnb":0, "abs":0, "tra":0}`

This should set speed/rpm on the cluster. If it doesn't then try lowering the baudrate in the sketch to 115200. See if that changes anything.

karrui, 2024-09-09:
Changing the baud rate did the trick! Thanks for the quick fix. Enjoying my dash.

karrui, 2024-09-09:
Another question: am I able to send the various light commands via SimHub? Stuff like main lights, high beam, etc. I do not see it in `SimhubGame.cpp`. What do I need to add for it? Basically using this for ETS2 and ATS.

r00li, 2024-09-09:
> Changing the baud rate did the trick! Thanks for the quick fix. Enjoying my dash.

Interesting. Well I am glad that worked.
I guess the serial interface is having some issues with higher speeds. Check the cable maybe.

> Another question: am I able to send the various light commands via SimHub? Stuff like main lights, high beam, etc. I do not see it in `SimhubGame.cpp`. What do I need to add for it? Basically using this for ETS2 and ATS.
> 
> Edit: looked in the source code. Will open a PR if it works!

Basically the way this works is - Simhub generates a simple JSON with parameters. Those are then read in SimhubGame.cpp and then applied to the game simulation object. So in theory you should be able to add those parameters easily. Main issue is getting them from Simhub - not sure if it supports all of them out of the box. I mainly didn't bother adding it myself, because I mostly play racing games which don't send this data.

karrui, 2024-09-09:
yup, ive updated the source code to do it. ETS2 sends the data! Tell me if you want me to submit a PR.

seems like a few cluster functions are not activated, such as brake lights. Are you planning to enable those?

r00li, 2024-09-09:
Yes, if you have the time, definitely open a PR and we will try and add it.

Brake lights? Not sure what you mean by that.
Functions that are available on each cluster really depend on the cluster itself and what CAN IDs I know. If you know CAN IDs that are not in the code then let me know and I will see if I can add them.

karrui, 2024-09-10:
Pushed a PR for the extra SimHub keys in #18.

Regarding the brake light signal, I was referring to these lights:

[image]

r00li, 2024-09-11:
Oh. The park brake indicator should work already. Though in a classic BMW fashion that appears to not show the indicator on all clusters. If I remember correctly there is a check in there and will either use one of the known IDs on some clusters, or it will use the check control on other clusters. Would need to find the proper ID that triggers it on other clusters. But as a workaround you might want to enable the check control way for now.

karrui, 2024-09-13:
That may not be the best image, but that is not the parking value, but the brake indicator that should show up every time you press the brake pedal.

r00li, 2024-09-13:
You are right that the parking indicator is in the center... the image really isn't the best. It's only gonna light up green in case of Auto H though. It's red (or orange) in case of parking brake activation.

As for the one on the right. That's a brake system warning indicator. It should definitely not light up every time you press the brake pedal. Actually... it should light up never 😄 . But if you want it to show up play with the check control CAN ID. I am pretty sure one combination of bits there will make it show up.

karrui, 2024-09-16:
ah! as you can see I do not drive in real life 😆. gotcha! not a big deal then.

closed 2024-09-16 by karrui