r00li/CarCluster #21: Could i use a shield insted of a mcp2515

milacic22, 2024-10-26:
Was wondering if i could connect a shield to an esp32 insted of a mcp2515?

r00li, 2024-10-26:
What do you mean by shield? Most CAN shields use an mcp2515 chip. Those will work out of the box, you just need to figure out the pinout of the shield so you can connect it correctly. Though I recommend the cheap little mcp2515 breakout boards. They are easier to wire up.

If you mean to use the internal CAN bus module of the ESP32... you will obviously still need a CAN transceiver chip. But you will also need to change the code to use that module. It is also doable, but will take a bit more effort.

milacic22, 2024-10-26:
https://www.aliexpress.com/item/32998255055.html?spm=a2g0o.productlist.main.5.4bc7U0OSU0OSzX&algo_pvid=e7e588d1-4107-42e5-bf87-89b34a4d36c9&algo_exp_id=e7e588d1-4107-42e5-bf87-89b34a4d36c9-2&pdp_npi=4%40dis%21BAM%2129.06%2120.92%21%21%2116.07%2111.57%21%402140c1c317299486112522018e119f%2166923485263%21sea%21BA%212619992915%21X&curPageLogUid=Ato4y2MlnLwa&utparam-url=scene%3Asearch%7Cquery_from%3A

I have this shield already and i was wondering if i could use that instaed of the mcp2515 module bc it will take a month to arrive and i am lazy to wait

r00li, 2024-10-26:
Yes... you can use it. You just need to figure out what the pinout for it is.

If you are in Europe then https://www.makershop.de/ has little mcp modules and such in stock usually, and will get them to you in about a week. So that is also an option if you can't get the shield working.

milacic22, 2024-10-26:
I already ordered it from aliexpress it will be here in about a month but i dont want to wait that long i will try to make it work thx for your help

closed 2024-10-26 by milacic22