r00li/CarCluster #22: Cluster škoda fabia MK2 5J0920810b

KRYSM4N, 2024-11-01:
Hello, I need help with the gauges from the Škoda Fabia II. I need to find out the CAN IDs or any guide on how to obtain them so I can activate them in SimHub and then play ETS2 while sending data from the game to the CAN bus MCP2515. I have been looking for these IDs for two weeks and have only found the ones for the tachometer and turn signals (0x280, 0x470). If anyone has all the CAN IDs for these gauges, I would greatly appreciate it. Thank you so much! <

r00li, 2024-11-01:
Hi!
I suggest you try my VW Polo 5 sketch. This is VW PQ platform so most of the stuff I guess should work. You might also try the Škoda Superb version - though I have a feeling Polo will be more similar.

kerrcz, 2024-11-09:
what pins are the CAN ones? i have dashboard from roomster so basicly the same stuff, in code i selcted pq25 and i have can low on 28 and can low on 29 and it is doing nothing

r00li, 2024-11-10:
Well, you will need to find a pinout. But usually they are the same on the PQ clusters, so 28 and 29 should probably be ok. Don't forget to connect the ignition and power pins though. Those are also needed. Try also swapping L and H around and enabling/disabling the CAN termination.

kerrcz, 2024-11-10:
i found them but i have another isue