r00li/CarCluster #24: Roomster cluster speed

kerrcz, 2024-11-10:
I have kinda working roomster cluster. RPM and indicators work but thats all i cant get working anything else any ideas to make atleast speed work?

kerrcz, 2024-11-10:
i have the wiring (in czech) i have rn just pin 8 and 7. i tried adding to that 5and 6 and the whol ecluster didnt do anything. it was on but i didnt react at all 
[image]

r00li, 2024-11-10:
Hmmm, looking at the pinout - speed signal is pin 6. That makes it sound like the speed isn't actually provided through CAN, but is instead done via a pin. That would explain some things. Also, I see some other stuff that should be on CAN apparently isn't.

kerrcz, 2024-11-10:
could i somehow simulate that with esp32?

kerrcz, 2024-11-10:
also that would explain why when i tried plugging in the speed into canbus the speed was jumping all around the place

r00li, 2024-11-10:
> could i somehow simulate that with esp32?

Yeah, it shouldn't be that hard. I think you should be able to do it with standard arduino PWM. Increasing the frequency will probably increase the speed. Start with something like 1Hz and go higher.

kerrcz, 2024-11-10:
and could it be possible to make it work with your code?

r00li, 2024-11-10:
Yes, it should. But you will of course need to do some modifications.

kerrcz, 2024-11-10:
if you hade some free time could you make those modifications? since i really dont know how to code

r00li, 2024-11-10:
Definitely not. Especially because you need the cluster there to actually check what is happening. The PWM stuff should be quite trivial - just search google for "ESP32 arduino PWM" and you will find some tutorials. But you will need to know some arduino basics first.

kerrcz, 2024-11-10:
it needs two pins so one that sends and ground or?

r00li, 2024-11-10:
I think the input is for the cluster to show stuff, output is for some other devices in the car that also need that information. So you would only need 1 pin.

kerrcz, 2024-11-10:
could you atleast try to write some really basic code for arduino nano so  that i can try it? and then i would try to continue by myself

r00li, 2024-11-10:
Well on an arduino Nano you can just try with tone first:
https://docs.arduino.cc/language-reference/en/functions/advanced-io/tone/

just write `tone(PIN, FREQUENCY);`. Obviously replace PIN with your PIN and FREQUENCY with the frequency that you want. As I said start with 1 and then increase it to 10, 100 and see what happens.

kerrcz, 2024-11-10:
i tried this and it didnt do anything https://docs.arduino.cc/tutorials/generic/secrets-of-arduino-pwm/