r00li/CarCluster #26: Does anyone know the pinout of this specific esp

milacic22, 2024-11-26:
I have this https://www.aliexpress.com/item/1005005704190069.html?spm=a2g0o.order_list.order_list_main.23.eeb218022HgUJA
Esp32 board i was wondering is the pinout the same/what is the pinout beacuse i cant get this to work 
Also do i need to do anything on the mcp board it self or just to connect it

milacic22, 2024-11-26:
I am trying to make this work on an f10 cluster for the last 4 hours and i simply cant i have tried everything

r00li, 2024-11-27:
Hi!
That looks like the same board that I am using - so a clone of devkit v1. Generally you should only need to connect it to the ESP32 and that should be it. For F10 you will also need to enable the termination resistor - the 2 pins on the header on the side of the board need to be connected together.

Now, one special thing about F10... Your ESP32 need to be plugged in and running before you plug in power to your cluster. In the serial monitor of the arduino IDE you should see "CAN BUS Shield init ok" before you plug in the cluster. That message indicates that ESP has started transmitting the CAN data.

milacic22, 2024-11-27:
Those are the little 2 ones next to  the can h and can l pins? And i still do everything like that now when i power it on it just displays the handbrake signal for like 2 sec and nothing afther that

milacic22, 2024-11-27:
it displays parking brake failure and turns off

closed 2024-11-27 by milacic22

r00li, 2024-11-27:
Since you close the issue, I am assuming that you managed to get it working?

milacic22, 2024-11-27:
Yea i did i am just the most incompetent person in the world the can low wasn’t plugged in i was 100% sure it was