r00li/CarCluster #27: Ardiuno code

baileyj123, 2024-11-29:
Hello there I got my cluster working , my cluster is in mph as I’m in the uk , the code is in Kmh , how can I change this so my clocks read right

baileyj123, 2024-11-30:
I am also getting put in P when connected to sim hub or playing a game but the speaker is very loud and makes a loud beeep noise is there a way to stop it from beeping ?

r00li, 2024-12-12:
The calculations and everything are in km/h. But all of the games output km/h when it comes to telemetry, so you shouldn't have any issues. I guess the only problem will be the web dashboard. But you should be able to mod the code there to display it in mph. You can go to WebDashboard.cpp, and you should be able to add logic to convert from/to mph to km/h there.

I am not quite sure I get your second question.