r00li/CarCluster #29: off topic mib2 car radio

dckiller51, 2024-12-19:
Hello,

Question I bought a mib2 car radio that I unlocked. I tried to connect the golf vii meter can directly to the car radio. The communication does not seem to work. Do I need a gateway or bcm between the two? sorry to pollute here with my question. For example, I would like to deactivate the winter tires via the car radio because the meter reached 200km/l and tells me so.

r00li, 2024-12-23:
Hi!
Sorry it took a while to respond...
Yes, the radio and the instrument cluster are generally on separate CAN buses - radio on infotainment CAN, and cluster on convenience CAN. A gateway will probably be required for them to talk properly together. I am guessing they work at different CAN speeds. You could of course also try making one using an Arduino and two CAN interfaces. But sadly I don't have much details about the infotainment CAN.

dckiller51, 2025-01-12:
Hello, I'm slowly progressing in my configuration. I received a "j533" gateway. I now have access with vcds. I also received your pcb (it's much cleaner and makes it easier to connect). 
I tested the configuration with 2 can buses (choice 99) but I'm facing a problem, I think that in this mode the system expects a key start. The meter lights up but remains on standby. 
1) How can I get around this? 
2) For the time on the meter on the Volvo project (Flybrick_S60R) it manages to retrieve the time from the computer, do you think we could integrate it into this project?

r00li, 2025-01-13:
1. Yes, the mode 99 is meant to be used with a BCM and a key and everything. In the .ino file here, go to this section:
```
#elif CLUSTER == 99
  // Golf 7 Passthrough mode
  MCP_CAN CAN2(VWMQB_PASS_CAN2_CS);  // Set CS pin

  long unsigned int canRxId2;
  unsigned char canRxLen2 = 0;
  unsigned char canRxBuf2[8];

  #include "src/Clusters/VW_MQB/VWMQBCluster.h"
  VWMQBCluster cluster(CAN2, ANALOG_FUEL_POT_INC, ANALOG_FUEL_POT_DIR, ANALOG_FUEL_POT_CS1, ANALOG_FUEL_POT_CS2, true);
  ClusterConfiguration defaultClusterConfig = cluster.clusterConfig();
#endif
```

Change the initialization line to this:
```
VWMQBCluster cluster(CAN2, ANALOG_FUEL_POT_INC, ANALOG_FUEL_POT_DIR, ANALOG_FUEL_POT_CS1, ANALOG_FUEL_POT_CS2, false);
```

This will revert the library back to sending all of the data. It will still pass data cluster -> gateway, but not other way around. If you need it to work the other way around as well, then you can go to `VWMQBCluster.cpp` and then in `handleReceivedData` remove the whole `if (passthroughMode == false)` block. If you need more fine grained control then search for `passthroughMode` and where it is used and you can enable/disable which IDs you want to be passed and which you want to send.

2. Problem is that I don't know how to actually set the date/time via CAN on MQB. It doesn't seem to be that straightforward. Otherwise getting the time either from Simhub or directly via internet from an NTP server shouldn't be difficult.

dckiller51, 2025-01-15:
I didn't succeed. I think it comes from my understanding of the wiring. 

1) Cluster connect to MCP2515 1 
Gateway (5,15) connect to MCP2515 2 

If it is this configuration do I have to put the two jumpers on the MCP2515 ? 

2) Cluster connect to gateway (8;18) 
Gateway (5,15) connect to MCP2515 1 

3) Cluster connect to gateway (8;18) 
Gateway (5,15) connect to MCP2515 2 

My other additions are: 

Radio on gateway (10;20) 
Climatronic on gateway (5;15)

I also wanted to help find additional codes but there again I did not find how it works. My idea was to use Xloader. If I understood correctly it sends all possible combinations in hex.

Exemple de cette méthode mais sans les explications malheureusement.

[https://random-science.com/2022/12/13/decoding-can-bus-data-on-a-car-instrument-cluster/](url)

dckiller51, 2025-01-18:
For now I found this by plugging a mcp2515 into a nano. When I change the time on the radio I have the following results. As a reminder the time of the car radio is transmitted to the cluster.

20h30

[image]

22h45

[image]

r00li, 2025-01-20:
Soon you will have a setup that is similar to mine: https://www.youtube.com/watch?v=VAaX5-TPA4Q

Basically I have it set up so that CAN1 of my board is connected to the gateway and CAN2 is connected to the cluster. So basically my board sits between the cluster and the gateway. That way I can send whatever I want to the cluster, but the rest of the modules work normally. Of course in my case I do have a BCM, so it is a bit different since I can get ignition for all modules without sending it myself.

I guess what you could try is connect the modules how they would be in the car to the gateway, but supply the ignition to the powertrain CAN on the gateway from the board. That should allow you to set the language from the radio.

I think the issue with language is that it is not a simple ID. A lot of the communication between radio and cluster is done via VWs BAP protocol. Which I have given up on ever decoding.

milacic22, 2025-01-20:
Hey i am kinda just throwing my self here sorry but i saw on ur yt chanel u have a f30 box setup I also want to make it just wondering do u have the code just for the idrive screen the accual unit and the idrive joystick so without the other modules

dckiller51, 2025-01-28:
@r00li 

Hello, I tried to modify the code to send the date. For now the time does not work when the cluster is connected.

Can you look at my modifications and help me on my errors (only the date in manual it works)?

No worries to compile but I think my is wrong.

I retrieve the date in simhub

'''
 ', "chh":' + format([DataCorePlugin.CurrentDateTime], 'HH') +
 ', "cmm":' + format([DataCorePlugin.CurrentDateTime], 'mm') +
 ', "cyy":' + format([DataCorePlugin.CurrentDateTime], 'yy') +
 ', "cmm":' + format([DataCorePlugin.CurrentDateTime], 'MM') +
 ', "cdd":' + format([DataCorePlugin.CurrentDateTime], 'dd') +
'''

[https://github.com/dckiller51/CarCluster/commit/8687514c41812bfd94853797e882fdaf9aa369a9](url)

dckiller51, 2025-02-06:
@r00li 
Hello, could you look at my modification of the code to send the time and date. I can't find my error, I've done lots of tests without success. Manually it works. I managed to add sliders to the webdash to facilitate testing.
My latest updates are on my fork.

MIGINC, 2025-02-07:
@dckiller51 I have had a look at your code. Have you tried setting `CAN.sendMsgBuf(DATE_ID, 0, 5, timeBuff);` to `CAN.sendMsgBuf(DATE_ID, 1, 5, timeBuff);`, as that's the flag to set extended CAN IDs.

Also, if anyone is interested in working together on reversing BAP, I have started a repo and will be organising a bunch of stuff I have worked out, think could lead to cluster \ radio menus being able to control settings in your sim setups. 
https://github.com/MIGINC/BAP_RE/

dckiller51, 2025-02-07:
> [[image]@dckiller51](https://github.com/dckiller51) I have had a look at your code. Have you tried setting `CAN.sendMsgBuf(DATE_ID, 0, 5, timeBuff);` to `CAN.sendMsgBuf(DATE_ID, 1, 5, timeBuff);`, as that's the flag to set extended CAN IDs.
> 
> Also, if anyone is interested in working together on reversing BAP, I have started a repo and will be organising a bunch of stuff I have worked out, think could lead to cluster \ radio menus being able to control settings in your sim setups. [MIGINC/BAP_RE](https://github.com/MIGINC/BAP_RE/)

Thank you, that was good. It works now. I need to fix the code a bit.

To participate I want to but I am a beginner in understanding Canbus and software. I think I have the equipment to do it.

I have the following equipment:

Raspeberry, Arduino Nano, Mega2560, ESP32, MCP2515....
For car : Cluster Golf7, Gateway 5Q0907530M, Climatronic, car radio Mib2 STD.

I actually think I need the BAP codes because my next thing is to have access to the Climatronic button.
I only have one that works. 

[image]

```
Button | ID | DLC | Data | Data Push
-- | -- | -- | -- | --
7 | 66E | 8 | E0 00 38 00 FE 00 00 00 | E8 00 38 00 FE 00 00 00
```

@r00li The good news is that it also works without a car radio or gateway so everyone can have the time and date automatically.

r00li, 2025-02-10:
> [@dckiller51](https://github.com/dckiller51) I have had a look at your code. Have you tried setting `CAN.sendMsgBuf(DATE_ID, 0, 5, timeBuff);` to `CAN.sendMsgBuf(DATE_ID, 1, 5, timeBuff);`, as [...]

That's a very interesting project. Very interested to see if you manage to figure out how the information about now playing song/radio gets sent to the cluster. Since that would enable us to display custom text on the cluster.