r00li/CarCluster #30: Handbrake indicator not working

martiplaysiuuun, 2024-12-23:
I am using a Polo 6R cluster (6R0920) and around half a minute after turning everything on the cluster chimes and the handbrake light starts flashing. I tried wiggling the wire, and the light turned off, so I put the cluster back, started playing and it did the same thing again. I completely changed the wire in charge but to no avail. I use wired mode with simhub and I was playing assetto corsa. I've also tried with beamNG.drive and Euro Truck Simulator 2. Any help would be appreciated.

https://github.com/user-attachments/assets/7faa8992-bb1b-4078-8b90-1c4f1b45627e

r00li, 2024-12-23:
Hmmm, try connecting the GND of your 12V power supply to the ground of the ESP32. I think I forgot to mention that in the wiring for the cluster.

martiplaysiuuun, 2024-12-24:
Sadly, still doesn't work.

https://github.com/user-attachments/assets/8ba3674e-351f-4914-9103-97a242498fb0

r00li, 2024-12-24:
You did wire the handbrake indicator pin to the ESP, right? So pin 25 of the cluster to pin D13 on the ESP?
If, yes, then try disconnecting it from the ESP and connect it directly to GND or directly to 12V - I can't remember which one turns the indicator off and which one turns it on. Let's see if this will fix your issue.

martiplaysiuuun, 2024-12-25:
Connecting pin 25 -> ESP's GND: turning the cluster on, the handbrake light is on as if I've pulled the handbrake, but after a little bit, it screams at me and light starts flashing.
Pin 25 -> 12V: Opposite. Light is off at first and then starts screaming. 
And yes, before testing, pin 25 of the cluster was connected to D13. Could it be my ESP32? I mean, it is from AliExpress, but everything else, without the gear counter works.

r00li, 2024-12-25:
Ok, since you connected it to GND/12V and it shows the same behavior then we know it's the cluster. So nothing wrong with the ESP.

The cluster being annoying with the handbrake on while you are moving is expected since that is what you would see in the car.

I remember seeing this problem once before, now that I think about it. I can't remember how I solved it though.

Do you by any chance have a resistor (anything in the line of 100 ohms to 10k ohm) around? If so, try connecting the handbrake pin via the resistor to either 3.3V of the ESP or the GND - basically creating a pull up/pull down state.

martiplaysiuuun, 2024-12-26:
Nothing. Same as before. Thank you very much for helping though. Oh, about the gear counter, my cluster doesn't show gears. Is it only for some games?

closed 2025-01-20 by r00li