r00li/CarCluster #31: ESP32 failing to boot

itsnotrin, 2024-12-24:
Hi, I flashed CarCluster and now my esp keeps rebooting with this error:

assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)

itsnotrin, 2024-12-24:
once i disable wifi it boots ^^

r00li, 2024-12-24:
Huh, that's an error I have not seen before. It certainly sounds like the issue with the ESP core.
Which ESP core version are you using? I am using 3.0.2
And which Arduino IDE version? I am using 2.3.2

itsnotrin, 2024-12-24:
I'm on Arduino IDE 2.3.4
I have a Lolin S3 Mini
esp32 core version 3.1.0

When it's fresh, the menu to select wifi etc is fine, it's once I point it at my actual wifi that I get the error.
After some research, it appears to be with ESPAsyncWebServer if that's any help

itsnotrin, 2024-12-24:
https://github.com/me-no-dev/ESPAsyncWebServer/issues/1451#issuecomment-2557361664

The solution seems to be to just downgrade my esp32 version until the repo maintainer fixes it. I'll message here if it works

itsnotrin, 2024-12-24:
I got it to work but now my bloody cluster won't work lol
Thanks :)

closed 2024-12-24 by itsnotrin

r00li, 2024-12-25:
Ah, ok. So an issue with the AsyncWebServer. Good to know. Will update it when the fix is available. Though I do eventually want to get rid of that dependency anyways.

Which cluster are we talking about? Anything that I can help with?

itsnotrin, 2024-12-25:
F10
It’s just to do with my ESP so that’s alright, I have a better one on the way

r00li, 2024-12-25:
Just note that F10 is a bit weird about booting up. The ESP needs to be sending CAN messages before the cluster is connected to power or it will simply refuse to turn on. So plug in the ESP, wait a few seconds then power on the cluster. And after uploading the code to ESP, unplug the cluster power, wait at least 10s and then plug it back in.