r00li/CarCluster #32: Help with Polo 6r instrument cluster

matheusmoreirainamine, 2025-01-03:
Hello, i'm from Brazil
 I recently bought a Polo 6R panel for my simulator. Well, I waited for him to arrive and I bought ESP32 and the necessary wires, but I need help with some things because I didn't understand what needs to be done. 
First of all, I would like help with the fuel level, I did not understand what is necessary for it to run. 
I would like help with Simhub too. Because I couldn't make the panel work there. I did all the necessary configuration in SimHub and when I open the game, the panel does not show the information. 
Sorry for my english, because i'm using Google Translator

matheusmoreirainamine, 2025-01-03:
I forget to say, i need help too with gear display, he doesn't work

matheusmoreirainamine, 2025-01-03:
Update: the simhub is working, my old cable doen't have data pin, i raplaced the cable and is working now

matheusmoreirainamine, 2025-01-04:
Hello, I'm here to update a few things. 
Well, I'm still in doubt about how to make the fuel marker run, but from what I understand I have to buy something for it to work, which at the moment is not feasible because of the expense I had in bringing the panel to Brazil, but in the near future I could buy, so I would like to know what is needed to make the marker work. 
The problem with Simhub was completely my mistake, since my cable didn't have the data wires needed for the ESP32 to connect to my PC, I bought a new cable, and it worked as it should. 
On the problem with the gear display, it has to do with the other topic that talked about the panel displaying SAFE and not displaying the gears. I haven't been able to see if there is a solution yet, but I will study better the solution. 
Okay, that said, I have another doubt, my friend is setting up a simulator based on the VW Jetta MK5 2008. Does anyone know if Is there any configuration that this panel works? The body is base PQ35

r00li, 2025-01-06:
Hi!
So, to simulate the fuel level on the Polo you need an X9C102 digital potentiometer chip and two 1k ohm resistors. 
The code is already there, you just need to wire it like this: https://github.com/r00li/CarCluster/blob/main/Misc/README_WIRING_JUMPERS.md
You can get those chips easily from Aliexpress: https://www.aliexpress.com/item/1005007022151823.html
If you are not using my PCB for wiring then you can use a small breadboard to wire everything.

As for the Jetta... if it is PQ35 then the code from Polo should work as well. But of course I haven't tested it with that cluster, so I can't say for certain.

matheusmoreirainamine, 2025-02-03:
Hello, i’m back
I’m tested with polo mode of your project, but i’m have some problems with uploading in the ESP
My ESP keeps rebooting, and show a error, i’m sending a picture of error, but in console shows reboot after some seconds, and keeps rebooting.
I’m belive my esp has some problem or i’m don’t know how to upload the sketch to esp, because he shows sometimes the error 2 in console

matheusmoreirainamine, 2025-02-04:
I'm found a topic with same error i have: 
assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)

And after this error, keeps rebooting.

I don't know why, but the ESP does not enter in download mode too

r00li, 2025-02-10:
Hi! 
The issue is the same as described here: https://github.com/r00li/CarCluster/issues/31

You need to install arduino esp32 core version 3.0.x and not 3.1.x. I will update the libraries to fix this as soon as I can, but until then please use the older core.

matheusmoreirainamine, 2025-02-14:
Oh, i remember this topic, i'm will test this, thanks
Do you want me to create a new topic for Jetta/Golf 2007 instrument cluster to help other peoples to search to a compatible instrument cluster?
I'm want try to adapt your project to work with Ford Fiesta MK7 (2015-2019), i have a lot of information of this instrument cluster, because i'm have this car in Brazil, i mapped some can id's, and i have a project from other user called Fereste. Do you authorize this attempt?

r00li, 2025-02-15:
> Do you want me to create a new topic for Jetta/Golf 2007 instrument cluster to help other peoples to search to a compatible instrument cluster?

What do you mean?

> I'm want try to adapt your project to work with Ford Fiesta MK7 (2015-2019), i have a lot of information of this instrument cluster, because i'm have this car in Brazil, i mapped some can id's, and i have a project from other user called Fereste. Do you authorize this attempt?

This project is open source. You can take it and modify the code however you like. If you want to add support for another cluster you can do that and open a PR.