r00li/CarCluster #34: Golf 7 fuel simulation

AmongUS987, 2025-01-19:
I have a question, do I have to buy x9c102p for the Golf 7 cluster or not?

dckiller51, 2025-01-20:
Hello, you need 2.

[https://github.com/r00li/CarCluster/blob/main/Misc/README_WIRING_VW_MQB.md](url)

r00li, 2025-01-20:
Indeed. For a Golf 7 you need two. 

Though I never quite finished the parameters for the simulation on the Golf. The code and wiring should work fine, just that you will need to tweak the values used for simulation.

AmongUS987, 2025-01-21:
> Indeed. For a Golf 7 you need two.
> 
> Though I never quite finished the parameters for the simulation on the Golf. The code and wiring should work fine, just that you will need to tweak the values used for simulation.

And do I have to change something in the code for a diesel cluster that ends the tachometer at 6?

dckiller51, 2025-01-21:
> And do I have to change something in the code for a diesel cluster that ends the tachometer at 6?

In CarCluster.ino line 33 you can define it.

AmongUS987, 2025-01-21:
> In CarCluster.ino line 33 you can define it.

So, just change #define MAXIMUM_RPM from 0 to 6?

dckiller51, 2025-01-21:
> > > > Indeed. For a Golf 7 you need two.
> > > > Though I never quite finished the parameters for the simulation on the Golf. The code and wiring should work fine, just that you will need to tweak the values used for simulation.
> > > 
> > > 
> > > And do I have to change something in the code for a diesel cluster that ends the tachometer at 6?
> > 
> > 
> > In CarCluster.ino line 33 you can define it.
> 
> So, just change #define MAXIMUM_RPM from 0 t 6.

I haven't tested it but I think you should put 6000.

AmongUS987, 2025-01-21:
> [@AmongUS987](https://github.com/AmongUS987) It works fine till 8k rpm

But my tachometer ends at 6k rpm

aslobodskoy, 2025-01-21:
@AmongUS987 w/o red zone till 8k?

dckiller51, 2025-01-22:
By default it is this line so you must put 6000.

Line 61 in VWMQBCluster.h

  config.maximumRPMValue = 8000;

r00li, 2025-01-27:
You just need to change the `#define MAXIMUM_RPM 0` to `#define MAXIMUM_RPM 6000`. It doesn't really change much though. This mostly just works as a limit. A lot of games don't actually output the maximum RPM, so only with the games that do you will get a bit better scaling.

closed 2025-01-29 by AmongUS987