r00li/CarCluster #35: Bmw f10 cluster mph

milacic22, 2025-01-20:
I have a bmw f10 cluster everything works the cluster is in km/h but for some reason the little display is showing speed in mp/h and the comsumption im mpg?

r00li, 2025-01-20:
That should be easy to solve. Send it:
`CanSend(0x291, 2, 18, 0, 0x00, 0x00, 0x00, 0x00, 0x00);`

There is a function in `BMWFSeriesCluster.cpp`called `updateLanguageAndUnits`. Uncomment the line above, then go to `updateWithGame` function and add a call to the `updateLanguageAndUnits();` under `sendDriveMode(game.driveMode);`. Upload the sketch, let it run. This should switch the cluster to km/h. Then remove the call and upload the code again - this is a call that you don't want to be making constantly - that is why it is commented out.

milacic22, 2025-01-20:
Thank you i aslo commented on a on another post regarding your f30 setup could you just look at that

closed 2025-01-20 by milacic22