r00li/CarCluster #37: BMW Series 1 F20 Support

davey422, 2025-01-25:
Would it be possible to add BMW Series 1 F20 Support? Functions like RPM and Fuel indicates are not working yet, however the Speedometer and the left/right indicator do work.

Thanks in advance!

r00li, 2025-01-27:
That is interesting. I would imagine that they would use the same F series IDs. For fuel indicator I do suggest that you try out using the Mini F55 option - it could be just that the mapping is wrong and the Mini should be closer to the correct mapping.

I did get some people mentioning that some F30 clusters also had issues with RPM, so I am guessing it is a similar problem. Most likely depends on the configuration of the cluster. But not sure what exactly. You can try finding the correct IDs, and we will see if we can figure out what is going on.

But I don't have any specific plans on adding it for now. I have a huge backlog of clusters that I want to work on, so I would rather add those first.

davey422, 2025-01-27:
Understandable, sadly the Mini F55 function didn't work either. if i figure something out i will get back to you with the solution, however I first have to learn a bit more about how a can bus works exactly.

CDFN, 2025-02-06:
Hi, I have F20 cluster (actually it's from X1 but it's exact same cluster) and it works flawlessly. I think you might have something wrong with your settings. 

[image]

r00li, 2025-02-10:
My feeling is that it has something to do with the cluster coding. Problem is that I don't know what exactly. Of course you could get a gateway module from one of the F series cars (I guess F10 ZGW is the easiest) and wire the diagnostic port and then use esys to play with the settings. But that is not a trivial thing to do.

CDFN, 2025-02-10:
I've done research on creating bench setup for coding clusters. You would need NBT head unit and FEM module. And chinesium adapter that will allow you to wire everything together. That comes down to 300-400USD so definitely not worth the buck for one time job. 

But I agree, this is possibly something to do with coding. Happy to share eeprom dump of mine (I believe coding is stored there), although you'd need programmer that will write 95128 EEPROMs (which aren't that easy to get hands on).

CDFN, 2025-02-14:
Actually, I had another F20 LCI cluster w/o extended display and this issue appears to actually exist. RPM gauge doesn't work, however it moves from "off" to "ready" when ignition is applied. Fuel gauge doesn't work too. Below attached what car the cluster is out of exactly.

[image]

r00li, 2025-02-15:
> I've done research on creating bench setup for coding clusters. You would need NBT head unit and FEM module. And chinesium adapter that will allow you to wire everything together. That comes down to 300-400USD so definitely not worth the buck for one time job.

Actually you don't need a HU. At least for coding a bkombi. Something like 6WA is a bit more complicated. Does your cluster have MOST connection? If not then you definitely don't need a HU. Actually I am pretty certain that you can code even clusters with MOST, though you might have some issues getting esys to detect it. 

Also a FEM is not required. An F10 ZGW should work. And those are around 10-15€. Those also have MOST connections, though I can remember if the MOST master is the HU or the ZGW.