r00li/CarCluster #38: Indicators not working in SimHub

AmongUS987, 2025-01-29:
I have a Golf 7 cluster and if I use WIFI connection all indicators work as they should, but when I connect ESP32 via SimHub the only thing that works from the indicators are the turn signals. How to fix it?

dckiller51, 2025-02-01:
1) check that you have selected the game you are playing. Well selected at each game change. 

2) checks that the game data arrives in the custom device. You click on edit you should see the data on the bottom line.

r00li, 2025-02-10:
Which indicators are you expecting to work, but don't?
Which game are you trying to play?
Keep in mind that a lot of the games don't actually provide telemetry data for those indicators. But really depends on the game.