r00li/CarCluster #40: Audi A3 Instrument Cluster Issues

ly200412, 2025-02-17:
I have an A3 instrument panel. I connected it according to your tutorial, and all the accessories were purchased as per your guide. The wiring has been checked and there are no issues, and the firmware is also installed correctly. However, when I power it on, the screen remains black and unresponsive. Interestingly, the panel works when I use the instrument startup platform. Additionally, the fuel level indicator is not responding. I am using the X9C102 module to drive the fuel sensor, but no matter how I debug it, it doesn't work. I hope you can help me with this. My English is not very good, so I would appreciate it if you could assist me.

r00li, 2025-02-17:
Which Audi A3 are we talking about? MQB based one? Or older PQ based?

> Interestingly, the panel works when I use the instrument startup platform

I don't quite understand what you mean by that.

> I am using the X9C102 module to drive the fuel sensor, but no matter how I debug it, it doesn't work. I hope you can help me with this

This one can be a bit tricky to get working. The trick is usually to unplug the cluster, set the fuel value to about 50% and then plug in the cluster. Then see if/how the fuel level indicator moves. After that you start moving the fuel level down and up and changing the mapping values until you get something that makes sense. If you have two fuel level pins on your cluster it makes sense to start with the primary one first and only then work on the secondary one.

ly200412, 2025-02-17:
This instrument is from the MQB platform. When I connect it using professional measuring tools, it works fine, but it doesn't start up when using ESP32. I'm a bit confused. Is the A3 startup message incorrect, or is there something wrong with the program? Also, how can I measure the A3 fuel gauge? What is its operating current and resistance? Isn't the ESP chip universal?

r00li, 2025-03-03:
So, what do you see in your arduino serial monitor? Is the CAN bus initialized successfully? 

A lot of the code for the MQB cluster in this projects was based from the Audi A3 logs, so generally at least ignition/speed/RPM should all work. 

Fuel gauge on the VW cluster is analog. You need an aditional digital potentiometer to simulate fuel. More information is here: https://github.com/r00li/CarCluster/blob/main/Misc/README_WIRING_JUMPERS.md#fuel-level-simulation-for-clusters-with-analog-fuel-level-indicators

ly200412, 2025-03-14:
Could you provide me with the complete CAN ID data for the cruise control indicator and lane-keeping indicator on the Audi A3 dashboard? These two indicators are not working, and I couldn't find the data controlling them in the code. Alternatively, some reference materials would also be helpful.

r00li, 2025-06-15:
Hi! 

> Can you share some Audi A3 CAN logs or links to such resources? I am interested in learning about CAN messages sent from Radio/Navi Headunit to investigate how I could push data to the instrument cluster with further information. ;-)

I sadly can't. The author of those logs has asked me not to share them, so I will respect that. So you will need some IDs first, then I can see if they are present there. Look at the BAP protocol. There is a high chance that this is what is actually being used for the communication here. But I am not sure. I could never really make any sense of this.

r00li, 2025-06-26:
Oh, that is really cool. I do wonder if BAP has changed much between PQ and MQB. I would kind of expect that it would be similar - though probably with different IDs being used.

r00li, 2025-06-26:
> I have a VW Sharan 7N2 at home which should be MQB as well. Idea is to build small CAN logger which I want to connect to the vehicle to log MQB data especially into direction of BAP. But I'm not sure where I need to connect to the CAN. OBD-2 port probably does not show the internal BAP communication as it is designed only for Diagnostics communication. That would mean I need to access the gateway somehow and I'm not sure if I really want to do that. Risk is to high that I damage something. Afterwards I have a detailed CAN log and a dead Sharan in my garage. ;-)

Yeah, you won't get the needed stuff on OBD. Gateway is one option, the cluster would probably be even better. Maybe also the headunit, but HU doesn't sit on the same bus as the cluster, and I am not sure how much magic is done by the gateway regarding these messages.

ly200412, 2026-11-02:
Can the ABS indicator and ESP indicator functions of the Audi A3 (MQB) instrument be added? Can you provide some assistance in implementing these two functions