r00li/CarCluster #46: Gear not showing on PQ35 (6R0920 860F)

martiplaysiuuun, 2025-03-17:
Everything is wired up correctly, even the fuel simulation works! Only thing that I don't have is the gear counter. Sone things I see is that every time I turn on the cluster, it says CHIAVE NON TROVATA once and then it disappears. That stands for key not found, right?[image]

[image]

[image]

r00li, 2025-04-15:
Regarding the gear display... I have a feeling this is related to the transmission type that was in the car and coding of the cluster. Because you are not the first one with this issue.

If you want to play around with it, I guess you could get a PQ platform gateway and one of the VCDS clones and try coding different options to the cluster. But that is just a theory.

As for the ignition... yes. I am not sending the ignition data in the custom script, so ignition will always be on. ETS2 seems to be the only game that actually supports that much telemetry. Pretty much all of the other games give you speed, gear display and that is pretty much it.

Generally if the game data is not received/game is not running you should get a door open warning, but that is it.

HONG18304006889, 2025-08-26:
Hello, may I ask how your fuel level gauge works? Do you need to add something else?