r00li/CarCluster #47: any chance to get this working with a PQ35 cluster with an arduino uno?

TrashCompactorACR, 2025-04-01:
basically what the title says

r00li, 2025-04-15:
Hi!
Generally you can in theory get the code to work on an Arduino Uno by simply removing all of the wifi/direct game connection code and the associated libraries. Can it be done? Yes. Do you want to? Probably not.

TrashCompactorACR, 2025-04-19:
sorry if my question is dumb why wont i want to?

r00li, 2025-04-19:
Mostly just because it is gonna take some effort and I am not even sure how good the result will be. It will probably work, but I also can imagine that some timings might be off since an Uno is much slower than an ESP32. I am not sure if it is worth spending the effort, when 10€ will get you an ESP32 board delivered.