r00li/CarCluster #48: wiring confusion

Dhruvit62442, 2025-04-19:
Hi wanted to ask about the 12 v power supply wiring in regards to ground connections as its shown that grounding should be done between cluster and power supply. 
Are we suppose to short ground connection of cluster and AC power supply getting for 12 V SMPS ? Just for information we are using indian AC power socket. It would help a lot for easier explanation of wiring of the whole setup (Car cluster is used from Volkswagen Polo India).

r00li, 2025-04-19:
For most clusters it generally doesn't matter. Especially the fully CAN ones. I think the BMW E60 was the only one where this did matter in my tests. 

But I would say that generally connecting the 12V GND and ESP32 GND together is a good idea. Especially if you are seeing that some of the digital signals for indicators are not working properly.

therandomguy-1413, 2025-05-17:
I am having the same issue. attaching a video reference for the same still not able to get the power supply to turn on the cluster. 
[https://drive.google.com/file/d/13n2FXJIYzag0Wbnb7VLZY2KgMCobwYgo/view?usp=sharing](url)

r00li, 2025-05-17:
So, if you press one of the buttons on the cluster (I can't remember which), does it light up? It should light up or at least show the odo. If if does, then your power supply is good, and you need to figure the CAN emulation part. If it doesn't, then the power supply/your wiring/your cluster are a problem.