r00li/CarCluster #49: Is there any easy way to reset/change the saved wireless network???

januszwoj, 2025-04-27:
Hi.
First of all I wanted to thank you for this project, it's really great. My son will be very happy when I give him a working car cluster.
I have a big problem, I wanted to change the wireless network but it is saved somewhere in the memory where I can't overwrite it (I have ESP32 WROOM Dev kit), probably the programs I'm trying to save don't reach these memory cells.
Is there an easy way to reset the wireless network settings?

r00li, 2025-04-27:
There are two ways...

1. In arduino IDE you have an option "Erase all flash before sketch upload" (in the tools menu). Enabled that and reupload the sketch. That should erase everything (including the wifi credentials)

2. In wifiFunctions.cpp file, before `bool res = wm.autoConnect(apName, apPassword);`add `wm.erase();` and reupload the sketch.

Whatever you do, reset the erase all flash option, or remove the added line and reupload the sketch before actually connecting to the new network.

januszwoj, 2025-04-28:
Thanks a lot for your help.
The first method worked immediately. I don't know how I missed that.
Thanks again.

closed 2025-04-28 by januszwoj