r00li/CarCluster #51: F10 Cluster Delay

BUJ5696, 2025-05-04:
I got my f10 cluster working with beamng using simhub, but few things does not seem to work and there is some significant delay. Basic stuff like RPM and Speed seems to be fine but there is some stutter, and side indicator lights are lagging. Serial baud rate is set to 921600. I wonder if it's hardware limitation of the can bus shield or esp32? Also would it be possible to get fuel emulation and other things such as light indicator working on this cluster?

r00li, 2025-05-17:
If you are using it with simhub are you using it with the license? Otherwise I know that the telemetry is a bit slow to update. Though even with that it shouldn't really cause too much lag. 

Fuel emulation should already work. But it probably will need some tweaks for your specific cluster. At least if your car was diesel. If it was petrol, then it might use different IDs, but I am not sure. But do play around with the mapping of the values. Most likely that could be the cause.

Light indicators for an F10 should work. But on beam this only means high beams - and only when connecting directly via wifi. Not via simhub. The issue is that simhhub doesn't have any game agnostic way of getting that data. And also beam doesn't really output anything other than high beams from telemetry.

BUJ5696, 2025-05-19:
Fuel emulation started to work randomly, and I'm using simhub without a license. About the lagging, everything seems to be fine with some slight delay. Turn signals, however, seems to have an irregular pulse, almost as if it is skipping beats. I will try it with WiFi later, and see if those issues persists