r00li/CarCluster #52: Reset wifi connection

trippo1, 2025-05-27:
Hello, 

First of all, I would like to congratulate you on a great project. Unfortunately, I have a problem with the wifi connection. I send the program to my esp, then log in to the network and set up my wifi. Then I see the connection below in the serial port monitor, which is interrupted every now and then. A reboot occurs and the situation repeats itself. Has anyone had this problem?

wm:STA IP Address: 192.168.1.114
Wifi connected...yeey :)

IP Address: 192.168.1.114

assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1854 (Required to lock TCPIP core functionality!)

Backtrace: 0x4008253c:0x3ffb1fd0 0x4008c3e9:0x3ffb1ff0 0x40092601:0x3ffb2010 0x40112717:0x3ffb2150 0x4011288d:0x3ffb2170 0x400d567c:0x3ffb2190 0x400e0286:0x3ffb21e0 0x400ebbd9:0x3ffb2200 0x400d3d8e:0x3ffb2240 0x400fa8a3:0x3ffb2270 0x4008d0da:0x3ffb2290

r00li, 2025-06-15:
Hi!
Are you using the correct version of the ESP32 arduino core? As the installation guide says, you need the arduino core version 3.0.x or lower. It will not work with versions 3.1.x or above. See here for details: https://github.com/r00li/CarCluster?tab=readme-ov-file#install-the-arduino-sketch-to-the-esp32

I still haven't had time to actually update this to work with the latest core.

r00li, 2025-12-19:
Ok, The project has now been updated to work with new arduino cores (3.3 was tested) so I am closing this issue. If you update the code and it is still not working correctly you can re-open the issue.

closed 2025-12-19 by r00li