r00li/CarCluster #53: Passthrough mode (CLUSTER 99) wiring

Milo629, 2025-06-12:
How should i wire this passthrough mode exactly? Do i need two MCP2515, one receiving from the CAN Gateway and the other one sending to the instrument cluster?

r00li, 2025-06-15:
Hi!
Why do you need the passthrough option? That is a pretty advanced option meant to be used with cluster, gateway and BCM (and everything else required to get the BCM into ignition). If you have that, then it is the correct option, otherwise you are looking for the "normal" option.

Using it does require two CAN interfaces. One interface goes to the output of the gateway that would normally go to the cluster (see wiring diagrams of the car), the other goes to the cluster. CAN interfaces share the SPI pins, except the CS and INT pins - should be clear from the code which goes where.

Milo629, 2025-06-16:
Thank you for your answer, did you have any issues using only one MCP2515 or why do you need this passtrough exactly?
If i look up the wiring diagrams on ELSA, for example the BCM (J519 - Onboard supply control unit) and the instrument cluster (J285 - Control unit in dash panel insert) also connects to the same "Convenience" CAN bus pin in the gateway (J533 - Data bus diagnostic interface)

LittleMonsterCurly, 2026-09-17:
Hello!
For a real car, it seems like the BCM needs to 'match' with the vehicle components. Do I need to worry about this in a test bench setup that only has the MIB radio and Cluster?

> The reason for the passthrough option is that I have a full sim rig - which includes the cluster, headunit, BCM, and other modules. The passthrough is there so I can inject the data into the cluster, while allowing the other modules to still talk to each other and the cluster. That is also why there are two CAN interfaces needed. One receives data from one side, filters out messages that are not needed (or I want replaced), the other then send the filtered/modified data to the cluster.
> 
> So, yeah... if you don't have a full set of modules then you want to use the option 3 not option 99.