r00li/CarCluster #54: Changing units in the golf 7 cluster

DhlDeliveryMan, 2025-06-17:
Hello!

I recently got a golf 7 cluster, and got it working with this project. However all the units are in imperial, and if possible i'd like them in metric. For speed this isnt a issue, as the speedometer is in both kmh and mph, however temperature, range and etc isn't.

Is there some way to change it ? Or are there some known CAN frames or any clues that with some tinkering could help me get closer to this goal ?

MIGINC, 2025-06-26:
The newer VW clusters get their time and units settings from the MIB units, this is via the BAP "Einheiten" channel can id (0x17331300) but is serialized with BAP, I am cleaning up a bunch on my notes and code on BAP to release, and this should help in getting that working.

r00li, 2025-06-26:
Yeah, VW made this complicated with BAP. Unlike BMW where this is just a normal CAN message, the VW one isn't as simple. 

Hopefully @MIGINC here can come up with a solution to this. I tried playing around with BAP a bit, but never really got anywhere.

If you are considering building a more complete sim rig then you can also try looking into getting a "semi-legit" headunit on ebay. If you search for MIB2-G you will probably find a few sellers which are selling a Chinese market VW headunit for relatively affordable price. They are good because they are not VIN locked. I have one of those in my sim rig and that one can change the cluster settings. You will probably also need a gateway if you go that route.